4ms/stm32mp1-baremetal #4: DDR3L memory

Staringlizard, 2021-10-01:
Nice work, and thanks for making this available!
Just a quick question about the RAM that I often see on the MP1 dev boards from ST. Is the RAM available and controlled automatically, or is this something that needs to be done to get that up and running?

danngreen, 2021-10-01:
Thanks! 

Yes, the DDR3 RAM needs to be initialized. Once it's up and running you can access it like any other RAM.

In this project, U-boot does the initialization. It's required to be done early in the startup process because the ROM bootloader is hard-coded to load just a small bit of code called the first-stage bootloader (FSBL) from the SD card (or from other sources, depending on the BOOT pins) into internal RAM (which has limited capacity).
The first-stage bootloader then initializes the DDR RAM and copies the second-stage bootloader (SSBL) from the SD card into DDR RAM. Finally it runs the SSBL. Our example apps are loaded by the SSBL.

Here's where the U-boot DDR driver code is:
https://github.com/4ms/stm32mp1-baremetal/tree/master/third-party/u-boot/u-boot-stm32mp1-baremetal/drivers/ram/stm32mp1

Ps: you could technically run the MP1 without DDR RAM, and just run everything in SYSRAM and the other internal RAM sections...

Staringlizard, 2021-10-03:
Thanks for clearing that out!

closed 2021-10-03 by Staringlizard