4ms/stm32mp1-baremetal #5: Ethernet example please

Undercroft, 2022-05-02:
Hi Dan,

thanks for providing these introductory projects. If possible, could you provide a very basic ethernet example? E.g maybe just hook up to a network with ARP and send or receive a UDP datagram.

danngreen, 2022-05-03:
Hi,
Based on a quick comparison between the H7 and MP1 ETH peripherals, it looks like the MP1 ETH peripheral features are a superset of the H7 ones. So, in theory it should be possible to use ST's H7 example projects as a starting point (similar to porting USB from H7 to MP1). I see several examples using lwip [such as these](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H735G-DK/Applications/LwIP). Probably a good place to start?

Are you using the Discovery board, or a custom board?

Undercroft, 2022-05-03:
Thanks. I'm not that familiar with the lwip examples so can't really comment. I have a STM32MP157D-DK1.