4ms/stm32mp1-baremetal #7: README should also describe debugging on Discovery board.

karelfv, 2022-05-28:
Hello,
thanks a lot for this project! I'll test that once the board is here, but to be honest I'm a bit confused by your highlighting of debug capability on the Octavo based board. Not sure if I read README correctly, but it feels like you completely ignore debug capability built into the STM32mp1 discovery board. Please let me know, have you used builtin ST-Link interface for A7 debugging on the board? Do you know about this capability? If it's new to you, then perhaps you can enhance the README. If for whatever reason it's not so great and have bugs, then this may also be mentioned there. Thanks!

danngreen, 2022-05-31:
Yes, the ST-LINK on the Discovery board is a viable way to debug. The README does talk about it, in Step 6: Debug Application (starting with “If you use OpenOCD with a USB cable connected to the ST-LINK micro-USB jack of the Discovery board…”) Does anything seem missing from that section?
But there are a few places earlier in the README that mention SWD on the OSD32 board, but not the built-in ST-LINK on the Discovery board. I can edit it to mention ST-LINK in those places. Thanks for bringing it to my attention, and good luck when your board arrives!

karelfv, 2022-05-31:
Indeed, hmm, I have to confess I just skimmed over the text quickly and described my feeling about. IMHO it was caused by requirements:
```
Optionally, a J-link or ST-LINK debugger (only if you use the OSD32 board)
```
No mention that ST-LINK is built into the DISCO already and by details below it which in length mention cheaper board debugging capability but is completely silent about disco.
Anyway, just nitpicks. Thanks for sharing of the great project!

danngreen, 2022-05-31:
I made some changes and pushed them to the main branch. Let me know what you think, I'll close this issue if you approve.

Feel free to share any tips you come up with as you use the on-board ST-LINK at any time. I've used it a little but most of my experience is with a direct SWD connection.

closed 2022-05-31 by danngreen