4ms/stm32mp1-baremetal #9: SD card formatting and partitioning

asg19660, 2022-08-14:
Hi Dan,

First of all I want to apologize as this issue is not really related to the bare metal code project per se.

Unfortunately I'm a beginner when it comes to Linux and I'm stumbling at the first hurdle - getting my SD card configured and loaded with the bootloader and application.

I am using 8GB SD cards (I've tried several, some brand new). I've tried using the format and partition scripts in the scripts directory but have also used other tools. I can see that the 4 partitions have been created however cannot format the 4th partition as FAT32 (error is too few blocks for a viable filesystem).

Even though I cannot seem to format the 4th partition I try to copy the bootloader (I'm using MP1-Boot) to the 1st and 2nd partitions. And then an example (in this case ctest) to the 3rd.

I'm using the STM Discovery board - it doesn't boot at all. No messages from the debug UART output.

Is there anything I need to do to a virgin SD card in order to prepare it for the above steps? Perhaps I am missing something silly?

I'm hoping you can help because I'm really looking forward to writing my application in bare metal (my eyes lit up when I saw your project because as you mention, there are next to no resources like this available).

Thanks in advance,
Andrew.

danngreen, 2022-08-14:
Hi Andrew,
So the “too few blocks for a viable filesystem” is probably because it’s an 8GB card, and there is a minimum size for FAT32 partitions, which depends on the FAT configuration. I don’t think this problem is causing the other issue (no booting, or at least no UART output on boot), since MP1-Boot doesn’t use the fourth partition. As long as the ‘dd’ commands that copy the bootloader are not giving an error, then you can ignore partition 4.

That said, I can think of two solutions: Use a 16GB card, or fiddle with the sgdisk flags to make it work.
I’ll play with the sgdisk command and see how small of a FAT32 partition we can use, and update the README.md.

For the other issue, I can think of a few things causing you to get no messages over the UART on boot:
- Are the BOOT switches on the Discovery board set to boot from SD Card?
- Are your terminal settings set to 115200, 8N1?
- Did you select the Discovery board in MP1-Boot’s ‘main.cc’ by uncommenting the line ‘namespace Board = STM32MP1Disco;’ before compiling?
- Are you using the STLINK micro-USB port for the UART connection?

If you still can’t get it to work, then you could verify if it’s booting by loading one of the examples that makes LEDs blink. Make sure to select the right board namespace before compiling, as always.
Another thing to try is to load the pre-built U-boot image onto your SD Card. Even without the app partition, U-Boot still sends messages on the UART, so you’ll be able to verify your UART connection and terminal settings.

asg19660, 2022-08-15:
Many thanks for your detailed reply Dan. I will try a larger SD card today to see if that fixes the FAT32 issue.

The confusing thing is that at the start of trying this had a working system. I had followed the instructions and programmed a new card (using U-Boot) and one of the examples and all was fine.

Then things started to go wrong. I was modifying one of the examples and then building it. Then dd'ing it to the card. Oddly when running the code on the board the old app was still running (I had changed the messages it was sending from the UART but it was still sending the original messages).

I verified that the app was indeed in the partition as I used a HEX viewer to view the card contents. So I was left totally confused as to how and why the old app was still there and running.

Then I decided to wipe the card completely and that's where is started to get even worse. At the same point I started to use MP1-Boot as in my application I need a very fast boot time.

To answer some of your questions - the boot switches are configured to boot from SD card. My terminal app is configured correctly. I have the disco board set in main.cc. I'm using the on board STLINK USB connected to my PC. Also I am using a PC with the latest version of Ubuntu desktop for development.

Thanks again, Andrew.

asg19660, 2022-08-15:
Dan,

Just to let you know I think all my problems have miraculously disappeared. I used a new 16GB card and all is now fine. Finally I have a reliable development cycle going so I can get on with the 'real' work of developing my application, which is a custom OS.

I've no idea why I had all the issues I had. I'm sure some can be put down to my minimal Linux experience.

Anyway, many thanks for your help and I'd like to say what you have shared with this project is great. I may well have some more questions in the future. If so hopefully more related to the actual CPU / baremetal...

Andrew.

closed 2022-08-15 by asg19660

danngreen, 2022-08-15:
Good to hear it!