4ms/stm32mp1-baremetal #10: MP1-Boot missing files

RainierLamers, 2022-09-14:
Config files are missing.

https://github.com/4ms/stm32mp1-baremetal/blob/master/bootloaders/mp1-boot/stm32disco_conf.hh

Does not exist

danngreen, 2022-09-14:
Hi,
stm32disco_conf.hh is in `examples/shared`:
https://github.com/4ms/stm32mp1-baremetal/blob/master/examples/shared/stm32disco_conf.hh

The Makefile includes this path, so it should build just by doing `make`. If you are not using make then you'll need to add the correct include paths to your build system (which you can see what they are by looking at the Makefile)

It's not the most intuitive place, I've been considering moving the board conf files to `boards/` dir at the root level.

danngreen, 2022-09-19:
Any updates on this? Are you able to find the file and include it in your project?

danngreen, 2022-09-29:
I merged a branch where I moved the shared directory from `/examples/shared/` to `/shared/`. This seems more natural since the files within are shared by all projects in `/examples` as well as `/bootloaders/mp1-boot`. Hopefully it makes usage more clear.

I update all projects Makefiles to use this new location for the SHAREDDIR.

Also, I added a Makefile in the /examples dir so you can easily build all examples. Just do:

```
cd examples
make all
```

I'm closing this because I hope this resolves the issue of not being clear where the common files are located. Please feel free to re-open this if you see a way to improve it further.

closed 2022-09-29 by danngreen