4ms/stm32mp1-baremetal #11: Comipling Source code error

SindhuVadde, 2022-09-28:
Hi,
Thanks for sharing the code. We have cloned the source code but when we run the following script scripts/build-u-boot.sh
 we are getting following error.
[image]
We have installed gcc version then also we are getting same error. How can i resolve this error.

danngreen, 2022-09-28:
The error message I see in your screenshot says `*** Your GCC is older than 6.0 and is not supported`. I believe that's referring to the Host GCC (not the cross-compiling GCC which is arm-none-eabi-gcc). 
What version of gcc is on your computer? Type this:

```
gcc -v
```

What does it say?
If it's earlier than 6.0, then you need to upgrade gcc on your host computer to a newer version, or you may just want to update your entire linux distribution (gcc 6.1 was released in 2016).

SindhuVadde, 2022-09-29:
Hi,
Thanks for the reply. We have downloaded [arm-gnu-toolchain-11.3.rel1-x86_64-aarch64-none-elf.tar.xz] from this **https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads**. For adding PATH Variable we refer the following **page.https://docs.oracle.com/cd/E19062-01/sun.mgmt.ctr36/819-5418/gaznb/index.html**
And Here is my path  /home/username/Downloads. I added path. Still it is showing the same error. Am i downloaded wrong tool chain or PATH setting procedure.

danngreen, 2022-09-29:
Ah, yes I see now. You downloaded aarch64-none-elf which is the wrong compiler. You need arm-none-eabi.

This is the one you need: arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi.tar.xz

Also, you need to add the path to the arm-none-eabi-gcc **binaries** to your PATH, not just the path to your download directory.
Please keep in mind that this has nothing to do with STM32MP1 or with this repo, it's really about learning bash commands and how to install software on your particular linux distribution. But I can try to help a little. You will need to learn some basic things about Linux and about using the command line, so keep reading basic Linux tutorials and basic command line usage.

Here is a relevant thread from the arm-gcc forum:
https://community.arm.com/support-forums/f/compilers-and-libraries-forum/52159/how-to-install-gnu-arm-embedded-toolchain

So, after you download and then unzip/untar the file, you need to add to $PATH the path to the actual file named exactly `arm-none-eabi-gcc`. For example if you downloaded and extracted the files into `/home/username/Downloads` then you would do this:

```
export PATH=/home/username/Downloads/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin:$PATH
```

Notice I added `bin/` to the end, because this is where the binaries live in the extracted files. Also notice I put the new path to the arm-none-eabi-gcc binary in the front of $PATH, rather than after it. This is to give priority to the newly installed binaries over any older versions you might have. 

Make sure that `/home/username/Downloads/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin` is an actual directory on your computer. I'm just guessing what it might be, so you must verify on your own system. If you type this:
```
ls -l /home/username/Downloads/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin
``` 
It should list the files in that directory, including one named `arm-none-eabi-gcc`. If it doesn't, then you need to figure out where you actually downloaded and extracted the files before adding this path to PATH

Again, I must stress strongly: you will need to learn basic command-line usage (also known as shell, which is typically bash) in order to make much progress. Here are some resources:

https://charmm-gui.org/?doc=lecture&module=unix&lesson=2
https://stackoverflow.com/questions/14637979/how-to-permanently-set-path-on-linux-unix
https://linuxhint.com/path_in_bash/
https://www.youtube.com/watch?v=rJMFxIbDe-g
https://www.youtube.com/watch?v=5tVofRTnJy4

danngreen, 2022-10-03:
Hi, @SindhuVadde just checking if you've made any progress.

danngreen, 2022-10-06:
I'm closing this, but feel free to re-open if you're still having troubles.