4ms/stm32mp1-baremetal #17: python error

Spitzbube, 2023-01-28:
I like this project very much because it is an entry point in bare metal programming of not only the A7 on the STM32MP1 but also other Cortex Cores. 
But trying this out initially I had a problem with the Python script creating the UImage header:

Creating uimg file
python ../../scripts/uimg_header.py build/main.bin build/a7-main.uimg
Traceback (most recent call last):
  File "../../scripts/uimg_header.py", line 37, in <module>
    image_name = bytes("stm32mp1-baremetal image", "ascii")
TypeError: str() takes at most 1 argument (2 given)
make: *** [../../shared/makefile-common.mk:127: build/a7-main.uimg] Error 1

After searching I found that there is a simple solution in shared/makefile-common.mk:127:
python has to be replaced with python3.

Greeting,
Thomas

danngreen, 2023-01-29:
Hi Thomas, 
That error probably means `python` is python 2 on your computer. What operating system and version are you using?
You're right that using python 3 will fix the problem. As far as the script and makefile, I'm not sure the best practice to ensure compatibility with most common operating systems, but after reading [PEP0394](https://peps.python.org/pep-0394/#for-python-script-publishers), it seems that using a python3 shebang at the top of the script is the preferred method to require python3. I'll update the makefile and python script, and add instructions in case that makes it break on other operating systems. 

That's great you got it running on the Avenger board. Can you share what changes you had to make? I'd be happy to add a board conf file for it, if you're willing to test it.

danngreen, 2023-03-30:
I changed `python` to `python3` in this commit: e25a57ee4056c05a2fe107ece6be9a2ceb6fc3ad
Hope it solves it. If you (or anyone) has issues still, feel free to re-open this

commit e25a57ee (2023-03-30): Explicitly use python3 for uimg_header

Closes #17
closed 2023-03-30 by danngreen