4ms/stm32mp1-baremetal #20: VS CODE IDE & OpenOCD debugging support

kamejoko80, 2023-04-30:
Hi @danngreen,

You know, vs code is a practical and lightweight IDE so I'd like to introduce how to add vs code IDE and how to configure the OpenOCD gdb server.

For a demonstration, I created [a repo](https://github.com/kamejoko80/stm32mp1-baremetal-1/tree/vscode), you can run vs code and open the folder "mp1-boot". 

This is just a sample for mp1-boot, you can organize the IDE for other programs if needed.

[image]

danngreen, 2023-05-01:
Great, this is helpful! Hopefully this will make it easier for others to debug and work with projects.

I got it working on my system with a few changes:

1) I'm not used to using VS Code, is it normal to have the toolchain binaries inside the project folder? If so, it would need to support more platforms, or support building the toolchain from source (which seems daunting), or perhaps support installation from xpack. Maybe the most simple way would be to tell the user to enter the path to their binaries into the `settings.json` file. To help support this, I changed it so that `TOOLCHAINS_DIR` is a full path containing the name of the arm-none-eabi installation dir. This way, the binaries can live anywhere on your machine and you can use any version you have installed, not just the xpack version 12.2. If there is a different way of handling it, that VSCode users will be familiar with?

2) I had to change the `targetProcessor` from 1 to 0. According to the [Cortex-Debug documentation](https://github.com/Marus/cortex-debug/blob/441af066c1168d2bd06fd962b0efefe7455c7892/debug_attributes.md?plain=1#L98), `targetProcessor` is a zero based integer index, so 0 would be the main core. Is it the same for you, or does your build of openocd index the cores differently? For me (using openocd 0.12.0), it will not debug and returns errors (as it should) if I have `targetProcessor` set to 1.

3) I think `.vscode/.cortex-debug.*.state.json` files should not be part of the repo since they just save the state of the debugger UI. I added them to the .gitignore and stopped tracking them.

4) I added another build target in `tasks.json` to build with optimizations turned on, and I gave both build targets more descriptive names.

Here's my changes (rebased from your fork):
https://github.com/danngreen/stm32mp1-baremetal/tree/vscode
I set the paths back to the ../../toolchains dir so it should just work for you.

kamejoko80, 2023-05-03:
Hi,

It's good to know that is working on your system.

1. Yes sure, you can use local toolchains, for flexibility to can define environment variables in settings.json.
2. In my case, only targetProcessor=1 works for me, here is the error log after switching to 0:
```
../toolchains/openocd/bin/openocd.exe -c "gdb_port 50000" -c "tcl_port 50002" -c "telnet_port 50004" -s ../toolchains/openocd/share/openocd/scripts -f "c:/Users/phuon/.vscode/extensions/marus25.cortex-debug-1.10.0/support/openocd-helpers.tcl" -f interface/stlink-dap.cfg -f target/stm32mp15x.cfg
Open On-Chip Debugger 0.10.0+dev-g32b1291e9-dirty (2023-04-22-14:47)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "stlink_swd". To override use 'transport select <transport>'.
cortex_a interrupt mask on
cortex_a interrupt mask on
cortex_a domain access control fixup on
cortex_a domain access control fixup on
50000
none separate
adapter speed: 10000 kHz
adapter_nsrst_assert_width: 200
adapter_nsrst_delay: 200
init
Info : Listening on port 50002 for tcl connections
Info : Listening on port 50004 for telnet connections
Info : STLINK V2J35S26 (API v2) VID:PID 0483:3752
Info : using stlink api v2
Info : Target voltage: 3.200000
Info : clock speed 10000 kHz
Info : SRST line released
Info : stlink_connect(connect)
Info : SWD DPIDR 0x6ba02477
Info : stm32mp15x.cpu0: hardware has 6 breakpoints, 4 watchpoints
Info : stm32mp15x.cpu1: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 50001 for gdb connections
Info : Listening on port 50000 for gdb connections
none srst_pulls_trst
Info : accepting 'gdb' connection on tcp/50000
Info : stm32mp15x.cpu2: hardware has 6 breakpoints, 4 watchpoints
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000008 msp: 0x00000100
Info : New GDB Connection: 1, Target stm32mp15x.cpu2, state: halted
Info : only breakpoints of two bytes length supported
Error: can't add breakpoint: resource not available
```
Can you share which OpenOCD you are using?

3. Yes, definitely.
4. Much appreciated for your improvement. Hopefully, you could find out some things in vs code configuration, cheer!