4ms/stm32mp1-baremetal #21: U-boot can't start example app, but mp1-boot can

nettercm, 2023-06-08:
Great stuff!

I have a STM32MP157F-DK2.
I built your u-boot, mp1-boot and hal_lib app.

When I follow the instructions to use u-boot to load the app, I get this error:

mmc0 is current device
55040 bytes read in 34 ms (1.5 MiB/s)
## Booting kernel from Legacy Image at c2000040 ...
   Image Name:   stm32mp1-baremetal image
   Created:      2023-06-08   4:36:08 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    55040 Bytes = 53.8 KiB
   Load Address: c2000040
   Entry Point:  c2000040
   Verifying Checksum ... Bad Data CRC
ERROR: can't get kernel image!
UBOOT>

Everything seems fine when I set up the sd card to use mp1-boot:

MP1-Boot

MPU clock: 648000000 Hz
Initializing RAM
Testing RAM.
Booted from SD Card
Loading app image...
Jumping to app
ABC
Hello from HAL Driver!
And hi to you too!
0
1
2
3
4
5

Any thoughts on why this might be happening?

danngreen, 2023-06-08:
Hmm, I see the same error on my end, I'll check it out. Thanks for reporting this!

danngreen, 2023-06-11:
OK, there was a bug in how the data size was calculated in uimg_header.py.
I just pushed the fix in 159a01dbb983bf8051e27dd62eace1673b05eef1
Thanks for finding that!

commit 159a01db (2023-06-11): Fix bad data size calculation in uimg_header script

Fixes #21
closed 2023-06-11 by danngreen