4ms/stm32mp1-baremetal #22: partition-sdcard.sh gives tons of errors

marcosartore, 2023-10-05:
Hi, I am following your tutorial step by step but am unable to fix these errors thrown when I execute your partition-sdcard.sh script:

Device /dev/sdb found, removing partitions
sudo sgdisk -o /dev/sdb

***************************************************************
Found invalid GPT and valid MBR; converting MBR to GPT format
in memory. 
***************************************************************

Non-GPT disk; not saving changes. Use -g to override.

Partitioning...
sudo sgdisk --resize-table=128 -a 1 -n 1:34:545 -c 1:fsbl1 -n 2:546:1057 -c 2:fsbl2 -n 3:1058:5153 -c 3:ssbl -N 4 -c 4:prog -p /dev/sdb

***************************************************************
Found invalid GPT and valid MBR; converting MBR to GPT format
in memory. 
***************************************************************

Setting name!
partNum is 0
Setting name!
partNum is 1
Setting name!
partNum is 2
Setting name!
partNum is 3
Disk /dev/sdb: 60526592 sectors, 28.9 GiB
Model: SD/MMC          
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): 215D93C7-D96B-49E6-B733-814245383042
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 60526558
Partitions will be aligned on 1-sector boundaries
Total free space is 0 sectors (0 bytes)

Number  Start (sector)    End (sector)  Size       Code  Name
   1              34             545   256.0 KiB   8300  fsbl1
   2             546            1057   256.0 KiB   8300  fsbl2
   3            1058            5153   2.0 MiB     8300  ssbl
   4            5154        60526558   28.9 GiB    8300  prog
Non-GPT disk; not saving changes. Use -g to override.

Formatting partition 4 as FAT32

mkfs.fat -F 32 /dev/sdb4
mkfs.fat 4.2 (2021-01-31)
mkfs.fat: unable to open /dev/sdb4: No such file or directory
umount: /dev/sdb: not mounted.
Done!

Thanks for helping,
Marco

danngreen, 2023-10-05:
Hmm, it sounds like the disk is MBR, not GPT so needs to be converted. Try this:
```
sudo sgdisk -g /dev/sdb
```

If that fails, `man sgdisk` suggests that the disk may be damaged. So try this:

```
sudo sgdisk -Z /dev/sdb
```

This will "zap" all partition data

marcosartore, 2023-10-06:
Thank you very much, the first command fixed the issue and I was able to get the serial messages with the expected final error.
Now I'm moving to install the arm-none-eabi-gcc toolchain ( downloaded the tar file, hopefully the good one among those tens of choices, unable to find installation instructions, a really slow starting for me....)
Marco

marcosartore, 2023-10-06:
If useful for anybody else, I've found this [link](https://lindevs.com/install-arm-gnu-toolchain-on-ubuntu) where it is explained how to install the arm-none-eabi-gcc toolchain

danngreen, 2023-10-06:
Thanks! I created a [Setup help guide](https://github.com/4ms/stm32mp1-baremetal/blob/master/docs/Setup.md) and included the link you gave. I also realized the format script is not needed, as the partition script can do the same, so I fixed that too.

closed 2023-10-06 by danngreen

marcosartore, 2023-10-07:
Thanks Dan, I want add a note which caused many troubles to me, despite it is TOTALLY independent from your repository and explanations. 
It is about the installation of the arm-none-eabi-gcc toolchain, which is needed to compile the examples. I followed the indications found at the following link: [https://lindevs.com/install-arm-gnu-toolchain-on-ubuntu](url)

I want to underline that it is probably due to my fault, as I installed everything with my login ('marco') and not as root. But many people can do like me and the following may help.
After installing the toolchain, I was however not able to "make" the examples as user 'marco', unless I issued:
`source /etc/profile`
After compilation I exited from the terminal window and all went OK.

I then switched off the computer and at the next restart the login 'marco' was blocked with an error regarding the entry 'arm-none-eabi-gcc' in /etc/profile. I had to build a boot ubuntu image onto an USB pendisk, boot from there, browse my Hard Disk, locate and remove the entry. Really ugly.
Still not tried to compile again, but now at leat my PC is recovered :-)

Marco