4ms/stm32mp1-baremetal #25: Building under linux, C++ projects are not building. C build fine

codewitch-honey-crisis, 2024-02-13:
I'm getting the strangest error trying to build the C++ projects.

```
rm -rf build
honey@Heavy:~/stm32mp1-baremetal/examples/usb_midi_host$ make
Building startup.s at -O3
Building main.cc at -O3
In file included from midi_host.hh:11,
                 from main.cc:13:
../../shared/drivers/interrupt.hh:14:30: error: 'function' in namespace 'std' does not name a template type
   14 |         using ISRType = std::function<void(void)>;
      |                              ^~~~~~~~
../../shared/drivers/interrupt.hh:4:1: note: 'std::function' is defined in header '<functional>'; did you forget to '#include <functional>'?
    3 | #include <functional>
  +++ |+#include <functional>
    4 |
../../shared/drivers/interrupt.hh:19:35: error: 'ISRType' has not been declared
   19 |         Interrupt(IRQType irqnum, ISRType &&func)
      |                                   ^~~~~~~
```

Now, <functional> is indeed include in interrupt.hh. I checked. So the header seems to be empty, but that can't be right?
So the error message is wrong.

Do you have any idea why this is happening?

danngreen, 2024-02-13:
I think that's it, it's due to a change in gcc 13 with regards to what "freestanding" means. This came up in another discussion here:

https://github.com/4ms/enosc/issues/1

One solution is, like you found, to use arm gcc v12.x or earlier.
Another solution is to remove the `-ffreestanding` flag. In that other project the only other change needed was to change the signature of `main` to `int main()`. I would expect it to be the same fix here.

codewitch-honey-crisis, 2024-02-13:
Whoops! Serves me right for not looking through the other issues more carefully. I've been at this for far too long. Imma take a break. Thank you. I got it building and I'm happy for now, and that means hopefully no more bugging you. This is great stuff. I've been trying to use xboot with allwinner A7 based SBCs to achieve similar but I never got anywhere with it. I don't read Chinese very well, so I only got so far. This code is really really fantastic. Breaks through a big wall for me. I owe you a keg of beer. =)

closed 2024-02-13 by codewitch-honey-crisis