4ms/stm32mp1-baremetal #28: Very interesting

Korporal, 2024-03-07:
This is very interesting, I'm interested in seeing what minimal environment one would need to be able to host the .Net CLR. It seems to me that currently such boards need Linux, Windows or perhaps some RTOS. Your detailed examples are giving me insights into the processor too.

danngreen, 2024-03-07:
The setup here works on MacOS, Linux, and Windows (using msys/mingw). You can use flash and debug from the command line, or you also can use Visual Studio Code and debug from that with the Cortex Debug extension.
I don't have VS but looking at VisualGDB, I would imagine it could be made to work. The STM32MP15x works with arm-none-eabi-gdb (same as many MCUs), so it's likely VisualGDB would work as well.

Korporal, 2024-03-07:
> The setup here works on MacOS, Linux, and Windows (using msys/mingw). You can use flash and debug from the command line, or you also can use Visual Studio Code and debug from that with the Cortex [...]

By the way, would these samples need adjusting for the STM32MP153 ?

danngreen, 2024-03-07:
No adjustments needed. I use the STM32MP153DAD in a device I'm developing, so all the examples here have been tested on that chip. As long as you're not trying to access features that aren't in the 157, it works fine.

closed 2024-04-09 by danngreen