4ms/stm32mp1-baremetal #35: Audio latency

zanellia, 2025-02-26:
Really cool project, thanks for sharing! What's the audio round-trip latency you get? How much of it is due to the codec and how much to the rest?

danngreen, 2025-02-27:
It can run < 1ms latency easily. The block size for the audio DMA is the biggest factor. We have a commercial device using this framework and it can go down to round-trip of 0.9us when you set the block size to 16 and sample rate to 96kHz. If latency was a higher priority than CPU efficiency, then it could go lower (I think I tried it with a block size of 4 for 0.2-0.3us latency once).

zanellia, 2025-02-27:
Pretty cool. Thanks for the quick reply! How difficult would it be to something similar on a board with a slightly more powerful CPU? I was thinking of Cortex A-76 or so. I see that you have work going on here https://github.com/4ms/rk3566-baremetal. Is the plan to do something similar to what you did here for the STM32MP1?

danngreen, 2025-02-27:
We're in the very early experimenting stages with the rk3566. Rockchip's documentation is very lacking, so it's not as easy to get various peripherals running as it was with the STM chip.